cnlohr/esplocalizer #2: Autoimporting KiCad pretty footprints from hardware/*.sch

brainstorm, 2017-01-06:
Hello @cnlohr,

Thanks much for publishing your projects as open hardware as well as the firmware, always inspiring stuff to learn from you.

I have been trying to reuse your `.sch` since I have a couple of modifications in mind to your design. 

Sorry if this sounds obvious (I'm a relatively new KiCad user), but how can I auto-import the footprints for pcbnew without manually importing each `.pretty` library/module? as the KiCad docs state, i.e:

https://github.com/jdunmire/kicad-ESP8266

I thought about giving this giant meta-pretty-repo a shot to fire it once and forget, but the ESP footprint doesn't seem to be there:

https://github.com/monostable/kicad_footprints

I understand you might be using your existing installation/footprints from past projects, but what would be your advice to semi-automate this process of importing footprints?

brainstorm, 2017-01-06:
I also meant including the `.pro`, `.lib`, etc files from KiCad like in the ESPUSB project:

https://github.com/cnlohr/espusb/tree/master/hardware/usb_8285

Also, totally understand if you don't want to release those yet before monetizing some boards a bit more through ebay, happy to start from scratch myself instead ;)

brainstorm, 2017-01-06:
Hold on a minute... you are adding LoRa (RFM95W radio) to a sister board, that is so cool!!! Exactly what I wanted to do for my own board!

[image]

Closing this issue since I'm going on my own with KiCad from scratch, but would really appreciate forking/refactoring esplocalizer's hardware (instead of dealing with `??` and unresolved symbols/components from the bare `.sch` file)... is there a kicad-utils script or similar that helps resolving/downloading those?

closed 2017-01-06 by brainstorm

cnlohr, 2017-01-09:
That seems odd... It should be able to pull the footprints from the cache.   Perhaps I forgot to include the cache?

cnlohr, 2017-01-09:
Looks like I did!  Fixed.

brainstorm, 2017-01-09:
Awesome, thanks Charles! The eeschema looks great now with the cached file (no more `??`'d components). The netlist can be generated without problems but it seems like `Cvpcb` is not liking the part assignments/mappings:

[image]
[image]

cnlohr, 2017-01-09:
I believe most of them are in my public repository here: https://cnlohr.net/pubsvn/electrical/kicad/

Did you tell KiCad to download all footprints locally?  If not, you should.  It should give you the common parts like SM0603 and SOT23-5

brainstorm, 2017-01-09:
Thanks Charles for following this up, super appreciated.

I found that svn a while later exploring, but then I switched to building my own board from scratch since I want to remove/change quite a few components and get more acquainted with KiCad.

Btw, if you ever move that repo to github, let me know, happy to include it to this semi-automated community repo, I've been using it recently, really handy:

https://github.com/monostable/kicad_footprints

cnlohr, 2017-01-09:
I might, but, it's one of those things I like to keep on my own because I know where everything I need is.