cnlohr/esplocalizer #3: Fails to start wifi

fmaurer, 2017-01-20:
New install following instructions, cloned esp82XX-basic example, successfully flash firmware with "make burn" and "make burnweb" (no errors), but doesn't start a wifi hotspot. Serial output on reset shows: 

```
esp82XX Web-GUI
Version: v0.1-7-g06742-dev - Build Thu Jan 19 20:05:33 PST 2017 with -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=5Loading Settings: af / 0 / 69 / 69
Settings Loaded: ESP_930A72 / Default
Opmode: 0
Boot Ok.
```

I'm not sure how to debug why the Opmode would be 0 (wifi off) for this example project. Any pointers would be really helpful, thank you!

On MacOS 10.12

cnlohr, 2017-01-20:
I am not sure how it got into opmode 0.  

Try pressing and holding the PGM (GPIO0) button down for about 5 seconds, then releasing.  It should reboot into SoftAP mode.

Charles

fmaurer, 2017-01-20:
You did it! Thank you a million :)  Had to hold closer to 10 seconds but after I let go PGM button wireless AP showed up and serial output:

```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 28272, room 16 
tail 0
chksum 0x85
load 0x3ffe8000, len 884, room 8 
tail 12
chksum 0x59
ho 0 tail 12 room 4
load 0x3ffe8380, len 2612, room 12 
tail 8
chksum 0xb2
csum 0xb2
don't use rtc mem data
rlN5
    esp82XX Web-GUI
Version: v0.1-7-g06742-dev - Build Thu Jan 19 20:05:33 PST 2017 with -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=5Loading Settings: af / 0 / 69 / 69
                Settings Loaded: ESP_930A72 / Default
                                                     Opmode: 2
                                                              Default SoftAP mode: "ESP_930A72":""
                                                                                                  Boot Ok.
                                                                                                          IGMP Joining: 0104a8c0 fb0000e0
```

I'm putting together Mac OS setup guide, will make note of this.

closed 2017-01-20 by fmaurer

cnlohr, 2017-01-20:
That would be really awesome!  Thanks!