cnlohr/esplocalizer #4: esplocalizer firmware-c example missing files

fmaurer, 2017-01-20:
Cloned esp82XX-basic example compiled and flashed to esp without issue. Everything works great.

Unable to do `make all` in /esplocalizer/firmware-c, compiler missing 'lib' files. What does libmain.a do and where can I get it :) ?

```
xtensa-lx106-elf-gcc: error: /Volumes/case-sensitive/esp-open-sdk/lib/libmain.a: No such file or directory
xtensa-lx106-elf-gcc: error: /Volumes/case-sensitive/esp-open-sdk/lib/liblwip.a: No such file or directory
xtensa-lx106-elf-gcc: error: /Volumes/case-sensitive/esp-open-sdk/lib/libssl.a: No such file or directory
xtensa-lx106-elf-gcc: error: /Volumes/case-sensitive/esp-open-sdk/lib/libupgrade.a: No such file or directory
xtensa-lx106-elf-gcc: error: /Volumes/case-sensitive/esp-open-sdk/lib/libnet80211.a: No such file or directory
xtensa-lx106-elf-gcc: error: /Volumes/case-sensitive/esp-open-sdk/lib/libwpa.a: No such file or directory
xtensa-lx106-elf-gcc: error: /Volumes/case-sensitive/esp-open-sdk/lib/libphy.a: No such file or directory
xtensa-lx106-elf-gcc: error: /Volumes/case-sensitive/esp-open-sdk/lib/liblwip.a: No such file or directory
xtensa-lx106-elf-gcc: error: /Volumes/case-sensitive/esp-open-sdk/lib/libcrypto.a: No such file or directory
xtensa-lx106-elf-gcc: error: /Volumes/case-sensitive/esp-open-sdk/lib/libpp.a: No such file or directory
```

fmaurer, 2017-01-21:
Copied /lib and /ld from here https://github.com/espressif/ESP8266_RTOS_SDK

```
$ make all
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -Os -I/Volumes/case-sensitive/esp-open-sdk/include -Iesp82xx/include -I. -Iesp82xx/fwsrc -Iuser -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50 -DVERSSTR='"Version: 5fb30-dev - Build Fri, 20 Jan 2017 18:35:48 -0800 with -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50"' esp82xx/fwsrc/uart.c esp82xx/fwsrc/esp82xxutil.c esp82xx/fwsrc/flash_rewriter.c esp82xx/fwsrc/http.c esp82xx/fwsrc/commonservices.c esp82xx/fwsrc/http_custom.c esp82xx/fwsrc/mdns.c esp82xx/fwsrc/mfs.c user/custom_commands.c user/user_main.c user/i2c.c user/promiscuous.c -flto -Wl,--relax -Wl,--gc-sections -nostdlib -L/Volumes/case-sensitive/esp-open-sdk/lib -L/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a /Volumes/case-sensitive/esp-open-sdk/lib/libmain.a /Volumes/case-sensitive/esp-open-sdk/lib/liblwip.a /Volumes/case-sensitive/esp-open-sdk/lib/libssl.a /Volumes/case-sensitive/esp-open-sdk/lib/libupgrade.a /Volumes/case-sensitive/esp-open-sdk/lib/libnet80211.a /Volumes/case-sensitive/esp-open-sdk/lib/libwpa.a /Volumes/case-sensitive/esp-open-sdk/lib/libphy.a /Volumes/case-sensitive/esp-open-sdk/lib/liblwip.a /Volumes/case-sensitive/esp-open-sdk/lib/libcrypto.a /Volumes/case-sensitive/esp-open-sdk/lib/libpp.a /Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a -T /Volumes/case-sensitive/esp-open-sdk/ld/eagle.app.v6.ld -B/Volumes/case-sensitive/esp-open-sdk/lib  -o image.elf
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: image.elf section `.text' will not fit in region `iram1_0_seg'
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: region `iram1_0_seg' overflowed by 3880 bytes
```

I read this had to do with SDK > 1.5.2, but I thought I solved this with esp-open-sdk/Makefile

#VENDOR_SDK = 2.0.0
VENDOR_SDK = 1.5.2

> e.g. with SDK versions greater than 1.5.2, Espressif changed the IRAM management

I'm not sure if coping /lib and /ld was the right thing to do or is even related...

cnlohr, 2017-01-21:
and /include.   Typically I keep the whole SDK together and rename the SDKs.

Indeed that is the issue, however, Espressif made a new SDK that should work again.   You can get a pre-release version of it from here https://github.com/cnlohr/swadges2017 - the file called angus_patch.

Charles

fmaurer, 2017-01-21:
- added /include folder
- downloaded and linked prerelease SDK in esp-open-sdk/Makefile, as seen here:

```
# Vendor SDK version to install, see VENDOR_SDK_ZIP_* vars below
# for supported versions.
VENDOR_SDK = 2.0.0
#VENDOR_SDK = 1.5.2

#...

VENDOR_SDK_ZIP_2.0.0 = REDUCED_ESP8266_NONOS_SDK_2.0.0_20160810_with_angus_patch.zip
VENDOR_SDK_DIR_2.0.0 = REDUCED_ESP8266_NONOS_SDK_2.0.0_20160810_with_angus_patch
#VENDOR_SDK_ZIP_2.0.0 = ESP8266_NONOS_SDK_V2.0.0_16_08_10.zip
#VENDOR_SDK_DIR_2.0.0 = ESP8266_NONOS_SDK_V2.0.0_16_08_10

#the rest of the file...
```

- run `make all `
- cd to /esplocalizer/firmware-c
- `make all` gives same iram error

I'm not understanding where the esp82XX-basic example differs so much from the esplocalizer/firmware-c to produce this behavior. Both project's user.cfg have:

`SDK          = /Volumes/case-sensitive/esp-open-sdk`

Just tried a `make burnweb` using angus SDK and esp82XX-basic produced no errors.

Could you elaborate on "keep the whole SDK together and rename the SDKs" ? Do you keep the esp-open-sdk inside of esplocalizer?

My directory structure looks like this /Volumes/case-sensitive :
- esp-open-sdk
- esp82XX-basic
- esplocalizer

fmaurer, 2017-01-22:
Fixed it:

First was my mistake forgetting command is `make burnweb` while in /esplocalizer/firmware-c

However, this isn't all, it produced the following error:
`Auto-detected Flash size: 32m
Running Cesanta flasher stub...

A fatal error occurred: Invalid head of packet ('\xd8')`

This was fixed by commenting out the following line in /esplocalizer/firmware-c/user.cfg

#FWBURNFLAGS  = -b 1000000

cnlohr, 2017-01-23:
Which dongle do you have?

fmaurer, 2017-01-23:
PL2303 TA based cable, using 2 level shifters for 3.3V and RX

https://www.amazon.com/gp/product/B00QT7LQ88/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

cnlohr, 2017-01-23:
Aah! Ok, I think that one works better at 1,500,000 baud.  Dunno why.

Charles

fmaurer, 2017-01-23:
In my infinite n00bness I have misunderstood many things, but learned much.

make burnweb and make burn accomplish different things. I was able to get burnweb to work as expected, but not `make burn`. This may explain why I couldn't get the fast acc CSV file to work: https://github.com/cnlohr/esplocalizer/issues/5 The esp hasn't been flashed to talk with IMU firmware code. All it has is the web GUI

It's the same `iram1_0_seg` error which I mistook as being fixed by using an older SDK. 

My /esp-open-sdk is compiled for 1.5.1, but that still produces same error. Next I'll try cnlohr's suggestion to keep SDK together and rename multiple SDKs.

```
$ make burn
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -Os -I/Volumes/case-sensitive/esp-open-sdk/include -Iesp82xx/include -I. -Iesp82xx/fwsrc -Iuser -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50 -DVERSSTR='"Version: 5fb30-dev - Build Mon, 23 Jan 2017 13:50:12 -0800 with -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50"' esp82xx/fwsrc/uart.c esp82xx/fwsrc/esp82xxutil.c esp82xx/fwsrc/flash_rewriter.c esp82xx/fwsrc/http.c esp82xx/fwsrc/commonservices.c esp82xx/fwsrc/http_custom.c esp82xx/fwsrc/mdns.c esp82xx/fwsrc/mfs.c user/custom_commands.c user/user_main.c user/i2c.c user/promiscuous.c -flto -Wl,--relax -Wl,--gc-sections -nostdlib -L/Volumes/case-sensitive/esp-open-sdk/lib -L/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a /Volumes/case-sensitive/esp-open-sdk/lib/libmain.a /Volumes/case-sensitive/esp-open-sdk/lib/liblwip.a /Volumes/case-sensitive/esp-open-sdk/lib/libssl.a /Volumes/case-sensitive/esp-open-sdk/lib/libupgrade.a /Volumes/case-sensitive/esp-open-sdk/lib/libnet80211.a /Volumes/case-sensitive/esp-open-sdk/lib/libwpa.a /Volumes/case-sensitive/esp-open-sdk/lib/libphy.a /Volumes/case-sensitive/esp-open-sdk/lib/liblwip.a /Volumes/case-sensitive/esp-open-sdk/lib/libcrypto.a /Volumes/case-sensitive/esp-open-sdk/lib/libpp.a /Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a -T /Volumes/case-sensitive/esp-open-sdk/ld/eagle.app.v6.ld -B/Volumes/case-sensitive/esp-open-sdk/lib  -o image.elf
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: image.elf section `.text' will not fit in region `iram1_0_seg'
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: region `iram1_0_seg' overflowed by 3880 bytes
```