cnlohr/esplocalizer #7: What is the power consumption in sleep mode?

harmverbeek, 2017-09-10:
Is it < 20 uA?
That would make it a really useful IoT device.

con-f-use, 2017-09-10:
There are different sleep modes in the esp8266, but yes, you can get in the ballpark of that but about everything on the chip will be disabled.

harmverbeek, 2017-09-10:
how about the usb uart chip and the leds?
are they also powered down?

con-f-use, 2017-09-10:
LEDs can be switched of in software or, in case of the power indicator LED desoldered on most ESP8266 modules - not all are the same. I don't know about the UART.

cnlohr, 2017-09-11:
THIS REMINDS ME!  I got permission to post a study I did on the sleeping options for the ESP on the esplocalizer!  It's really rough and totally not publish ready but here is a link to the google drive and some of the code I used.

https://drive.google.com/open?id=0B-rO-WAvEtVBU1FSOVRKZ3Npc1U

TL;DR:
 * Deep sleep on esplocaizer is 0.9mA
 * Sleeping, waiting for a GPIO wakeup is 1.8mA
 * Disconnected-from-wifi @pvvx light-sleep or as I call it "crazy sleep" where the processor is still running but not very utilized is 3.5-5.5mA depending on work load.  Thanks @pvvx you are one awesome dude.
 * Wifi on, light sleep ~12mA
 * Full bore: 72mA

cnlohr, 2017-09-11:
The real killer for suuuper tiny power consumption on this board is a combination of the regulators, and static uses.  I don't actually know how much the LEDs use, but it's not much.

harmverbeek, 2017-09-11:
0.9mA is 90 times too much for a battery powered sensor ;)
That's the problem with all sensor boards that include a USB UART chip, regulators and leds.
You have got to find a way to at least switch off the UART chip when not used.

cnlohr, 2017-09-12:
This board does not include the uart chip on it.  USB on here is strictly for power.  It's everything else on-board, including the lithium ion charging circuit.